eez-open/eez-framework #3: How to use eez-framework in Zephyr RTOS

nicogou, 2024-05-08:
Hi,
Using EEZ Studio I have designed a simple UI that uses flows. Before going too much further I wanted to try to incorporate this UI into my project that is based on Zephyr RTOS (nRF Connect SDK more accurately). What is the best way to do this ? I am not super familiar with Cmake, so I may be missing something obvious.
I first thought about using this repo as a reference : https://github.com/eez-open/zephyrproject-rtos-eez-lvgl/tree/main but it seemed outdated.
Then I tried to follow the instructions in this repo's main branch readme regarding modifications of CMakeLists.txt files, but I get errors when compiling eez-framework files.

```
Building app
C:\WINDOWS\system32\cmd.exe /d /s /c "west build --build-dir c:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app/build_shield c:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app --pristine --board xiao_ble_sense --no-sysbuild -- -DNCS_TOOLCHAIN_VERSION=NONE -DBOARD_ROOT=c:/users/nicol/documents/xiao-round-display/xiao-round-display.git -DCACHED_CONF_FILE=c:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app/prj_shield.conf -DDTC_OVERLAY_FILE=c:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app/boards/xiao_ble_sense.overlay -DSHIELD=seeed_xiao_round_display"

-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app
-- CMake version: 3.21.0
-- Found Python3: C:/ncs/toolchains/cf2149caf2/opt/bin/python.exe (found suitable version "3.9.13", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: C:/Users/nicol/Documents/xiao-round-display/zephyr/.cache
-- Zephyr version: 3.5.99 (C:/Users/nicol/Documents/xiao-round-display/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: xiao_ble_sense
-- Shield(s): seeed_xiao_round_display
-- Found host-tools: zephyr 0.16.5 (C:/ncs/toolchains/cf2149caf2/opt/zephyr-sdk)
-- Found toolchain: zephyr 0.16.5 (C:/ncs/toolchains/cf2149caf2/opt/zephyr-sdk)
-- Found Dtc: C:/ncs/toolchains/cf2149caf2/opt/bin/dtc.exe (found suitable version "1.4.7", minimum required is "1.4.6") 
-- Found BOARD.dts: C:/Users/nicol/Documents/xiao-round-display/zephyr/boards/arm/xiao_ble/xiao_ble_sense.dts
-- Found devicetree overlay: C:/Users/nicol/Documents/xiao-round-display/zephyr/boards/shields/seeed_xiao_round_display/seeed_xiao_round_display.overlay
-- Found devicetree overlay: c:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app/boards/xiao_ble_sense.overlay
node '/soc/spi@40023000/gc9a01@0' compatible 'galaxycore,gc9x01x' has unknown vendor prefix 'galaxycore'
-- Generated zephyr.dts: C:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app/build_shield/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app/build_shield/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: C:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app/build_shield/zephyr/dts.cmake
Parsing C:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app/Kconfig
Loaded configuration 'C:/Users/nicol/Documents/xiao-round-display/zephyr/boards/arm/xiao_ble/xiao_ble_sense_defconfig'
Merged configuration 'c:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app/prj_shield.conf'
Configuration saved to 'C:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app/build_shield/zephyr/.config'
Kconfig header saved to 'C:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app/build_shield/zephyr/include/generated/autoconf.h'
-- Found GnuLd: c:/ncs/toolchains/cf2149caf2/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe (found version "2.38") 
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/ncs/toolchains/cf2149caf2/opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc.exe
CMake Warning at C:/Users/nicol/Documents/xiao-round-display/zephyr/subsys/usb/device/CMakeLists.txt:22 (message):
  CONFIG_USB_DEVICE_VID has default value 0x2FE3.

  This value is only for testing and MUST be configured for USB products.

CMake Warning at C:/Users/nicol/Documents/xiao-round-display/zephyr/CMakeLists.txt:862 (message):
  No SOURCES given to Zephyr library: drivers__display

  Excluding target from build.

-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app/build_shield
-- west build: building application
[1/637] Building CXX object src/display/ui/eez-framework/CMakeFiles/eez-framework.dir/src/eez/core/action.cpp.obj
FAILED: src/display/ui/eez-framework/CMakeFiles/eez-framework.dir/src/eez/core/action.cpp.obj 
C:\ncs\toolchains\cf2149caf2\opt\zephyr-sdk\arm-zephyr-eabi\bin\arm-zephyr-eabi-g++.exe -DEEZ_FOR_LVGL -DLV_LVGL_H_INCLUDE_SIMPLE -isystem C:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app/src/display/ui/eez-framework/./src -isystem C:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app/src/display/ui/eez-framework/./src/eez/libs/agg  -MD -MT src/display/ui/eez-framework/CMakeFiles/eez-framework.dir/src/eez/core/action.cpp.obj -MF src\display\ui\eez-framework\CMakeFiles\eez-framework.dir\src\eez\core\action.cpp.obj.d -o src/display/ui/eez-framework/CMakeFiles/eez-framework.dir/src/eez/core/action.cpp.obj -c C:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app/src/display/ui/eez-framework/src/eez/core/action.cpp
In file included from C:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app/src/display/ui/eez-framework/src/eez/core/action.cpp:13:
c:\users\nicol\documents\xiao-round-display\xiao-round-display.git\app\src\display\ui\eez-framework\src\eez\core\action.h:15:10: fatal error: lvgl.h: No such file or directory
   15 | #include "lvgl.h"
      |          ^~~~~~~~
compilation terminated.
[2/637] Building CXX object src/display/ui/eez-framework/CMakeFiles/eez-framework.dir/src/eez/flow/components/label_in.cpp.obj
FAILED: src/display/ui/eez-framework/CMakeFiles/eez-framework.dir/src/eez/flow/components/label_in.cpp.obj 
C:\ncs\toolchains\cf2149caf2\opt\zephyr-sdk\arm-zephyr-eabi\bin\arm-zephyr-eabi-g++.exe -DEEZ_FOR_LVGL -DLV_LVGL_H_INCLUDE_SIMPLE -isystem C:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app/src/display/ui/eez-framework/./src -isystem C:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app/src/display/ui/eez-framework/./src/eez/libs/agg  -MD -MT src/display/ui/eez-framework/CMakeFiles/eez-framework.dir/src/eez/flow/components/label_in.cpp.obj -MF src\display\ui\eez-framework\CMakeFiles\eez-framework.dir\src\eez\flow\components\label_in.cpp.obj.d -o src/display/ui/eez-framework/CMakeFiles/eez-framework.dir/src/eez/flow/components/label_in.cpp.obj -c C:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app/src/display/ui/eez-framework/src/eez/flow/components/label_in.cpp
In file included from C:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app/src/display/ui/eez-framework/src/eez/flow/components/label_in.cpp:15:
c:\users\nicol\documents\xiao-round-display\xiao-round-display.git\app\src\display\ui\eez-framework\src\eez\flow\hooks.h:21:10: fatal error: lvgl.h: No such file or directory
   21 | #include "lvgl.h"
      |          ^~~~~~~~
compilation terminated.
[3/637] Building CXX object src/display/ui/eez-framework/CMakeFiles/eez-framework.dir/src/eez/flow/components/label_out.cpp.obj
[... 20 lines cut ...]
   57 |         VALUE_TYPES
      |         ^~~~~~~~~~~
[9/637] Building CXX object src/display/ui/eez-framework/CMakeFiles/eez-framework.dir/src/eez/flow/components/is_true.cpp.obj
FAILED: src/display/ui/eez-framework/CMakeFiles/eez-framework.dir/src/eez/flow/components/is_true.cpp.obj 
C:\ncs\toolchains\cf2149caf2\opt\zephyr-sdk\arm-zephyr-eabi\bin\arm-zephyr-eabi-g++.exe -DEEZ_FOR_LVGL -DLV_LVGL_H_INCLUDE_SIMPLE -isystem C:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app/src/display/ui/eez-framework/./src -isystem C:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app/src/display/ui/eez-framework/./src/eez/libs/agg [... cut ...]
My relevant CmakeLists.txt file is available on my project repo : https://github.com/nicogou/xiao-round-display/blob/eez-ui/app/src/display/ui/CMakeLists.txt
I am using a Xiao BLE Sense board, with a Round Display for Xiao shield. Note that the display driver and shield are not available in current nrf Connect SDK version, but they are available on the main zephyr RTOS branch. I manually copied them to my project's Zephyr folder.
Let me know if you need any more information.

Thanks for your help!
Best regards,
Nicolas Goualard

mvladic, 2024-05-08:
To fix these errors:

```
error: expected identifier before ',' token
   57 |         VALUE_TYPES
      |         ^~~~~~~~~~~
```

pull the latest version of eez-framework which I just committed - there was a syntax error which I committed recently, you are the first one who noticed this, thanks!

As for these errors:

```
fatal error: lvgl.h: No such file or directory
   15 | #include "lvgl.h"
      |          ^~~~~~~~
compilation terminated.
```

you can try to fix it by adding `include_directories(<path to lvgl directory>)` (for example `include_directories(lvgl)`, or wherever lvgl directory is in your case) line in your CMakeLists.txt. Or, maybe Zephyr has some dependency management setup, in which case you should setup that eez-framework depends on lvgl.

nicogou, 2024-05-08:
Thanks for the quick reply and fix!
I made the changes you suggested, it seemed to work! However it looks like I don't have enough flash on my device in order to run the code.
 
```
[634/639] Linking CXX executable zephyr\zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map C:/Users/nicol/Documents/xiao-round-display/xiao-round-display.git/app/build_shield/zephyr/zephyr_pre0.map 
cmd.exe /C "cd . && C:\ncs\toolchains\cf2149caf2\opt\zephyr-sdk\arm-zephyr-eabi\bin\arm-zephyr-eabi-g++.exe  -gdwarf-4 @CMakeFiles\zephyr_pre0.rsp -o zephyr\zephyr_pre0.elf  && cmd.exe /C "cd /D C:\Users\nicol\Documents\xiao-round-display\xiao-round-display.git\app\build_shield\zephyr && C:\ncs\toolchains\cf2149caf2\opt\bin\cmake.exe -E true""
c:/ncs/toolchains/cf2149caf2/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe: zephyr\zephyr_pre0.elf section `rodata' will not fit in region `FLASH'
c:/ncs/toolchains/cf2149caf2/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe: region `FLASH' overflowed by 48052 bytes
c:/ncs/toolchains/cf2149caf2/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe: src/display/ui/eez-framework/libeez-framework.a(value.cpp.obj): in function `eez::getVar(short)':
value.cpp:(.text+0x46f0): undefined reference to `native_vars'
c:/ncs/toolchains/cf2149caf2/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe: src/display/ui/eez-framework/libeez-framework.a(value.cpp.obj): in function `eez::setVar(short, eez::Value const&)':
value.cpp:(.text+0x483c): undefined reference to `native_vars'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\ncs\toolchains\cf2149caf2\opt\bin\cmake.EXE' --build 'c:\Users\nicol\Documents\xiao-round-display\xiao-round-display.git\app\build_shield'
```

I'll see if anything can be done about that, and if not I'll try to implement the UI without EEZ flows and still achieve what I want to do.

closed 2024-05-08 by nicogou

nicogou, 2024-05-08:
Oh I forgot to mention one thing though: there was an issue with the loadScreen function being declared twice, first in the ui.h file generated by EEZ Studio, and also in the lvgl_api.h file in eez-framework\src\eez\flow.
As these two were doing very similar things I removed the function from the generated ui.h file in order to keep compilation going, but I thought I'd mention it here.

mvladic, 2024-05-09:
> Oh I forgot to mention one thing though: there was an issue with the loadScreen function being declared twice, first in the ui.h file generated by EEZ Studio, and also in the lvgl_api.h file in eez-framework\src\eez\flow.
As these two were doing very similar things I removed the function from the generated ui.h file in order to keep compilation going, but I thought I'd mention it here.

This shouldn't be issue anymore as all LVGL templates (https://github.com/eez-open/eez-project-templates) and examples (https://github.com/eez-open/eez-project-examples) are recently corrected.

Luke-Shepp, 2024-05-13:
@nicogou did you find a solution for the native_vars issue? I'm having the same. https://github.com/eez-open/eez-framework/issues/4