eez-open/eez-framework #4: Undefined reference to `native_vars'

Luke-Shepp, 2024-05-13:
I'm trying to build eez-framework on [Circle RPI](https://github.com/rsta2/circle/). (I've got an eez lvgl project building and running great, I'm just trying to add in flow support now)

I'm getting this error during linking; **undefined reference to `native_vars'**

```
  LD    kernel8-32.elf
arm-none-eabi-ld: ../eez-framework/src/eez/core/value.o: in function `eez::getVar(short)':
../eez-framework/src/eez/core/value.cpp:1389:(.text+0x1978): undefined reference to `native_vars'
arm-none-eabi-ld: /../eez-framework/src/eez/core/value.cpp:1389:(.text+0x197c): undefined reference to `native_vars'
arm-none-eabi-ld: /../eez-framework/src/eez/core/value.o: in function `eez::setVar(short, eez::Value const&)':
/../eez-framework/src/eez/core/value.cpp:1420:(.text+0x3750): undefined reference to `native_vars'
arm-none-eabi-ld: /../eez-framework/src/eez/core/value.cpp:1420:(.text+0x3754): undefined reference to `native_vars'
make: *** [kernel8-32.img] Error 1
```

I've done a full search through eez-framework, lvgl, etc, and can't see a definition of native_vars beyond `extern native_var_t native_vars[];` in `src/eez/core/vars.h`

To help debugging the issue further - where should native_vars come from? Should it be from the eez-studio generated code?

mvladic, 2024-05-13:
In case if missing, put `//${eez-studio LVGL_NATIVE_VARS_TABLE_DEF}` in your project `flow_def.c` file template. It should be already there if project is created with the latest EEZ Studio.

[image]

Luke-Shepp, 2024-05-13:
Ah, I'm missing flow_def completely. I suspect that's because I've started with a non-flow project and enabled flow on it.

mvladic, 2024-05-13:
With these file templates we got some flexibility in the sense that the user can put his file among them or add something to the existing ones, but as you can see this can sometimes cause problems. We need to see how to make this more robust, for example, it can be displayed as an error if one of the mandatory directives such as `//${eez-studio LVGL_NATIVE_VARS_TABLE_DEF}` is not found in the file templates, without which the build does not go through.

Luke-Shepp, 2024-05-13:
Resolved by updating file templates to that of the LVGL with Flow project baselines, thank you!

closed 2024-05-13 by Luke-Shepp