eez-open/eez-framework #9: Does anyone have a working PlatformIO project for the CYD

PacciK, 2024-07-10:
I'm stuggling trying to make a first working project using eez-flow fomake a simple page one/two load with buttons as a starter.

All forms of errors coming up, including `src/ui.h:4:10: fatal error: vgl.h: No such file or directory`

Does anyone have a simple starter project working example for the [CYD2USB ](https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/blob/main/cyd.md)
Thanks a million.

PS. This is not an issue - ( but there's no discussion section , so I had to write here )

jcuypers, 2024-07-10:
it should be lvgl.h not vgl.h

PacciK, 2024-07-10:
@jcuypers - ThankYou !!  that **vgl** reference was due to my mistake in opening the nth attempt ( and enditing the LVGL include path in **Settings/Build** in EEZ Studio.
You gave me some hope to continue. Thanks.

jcuypers, 2024-07-10:
No worries we have all been there.

mvladic, 2024-07-10:
Can you share your project on github so I can try to build it?

PacciK, 2024-07-10:
@mvladic - Martin, thanks for offering to have a look.

Github was complaining at all the files to upload, so I zipped the 
a) Platform IO folder - I removed the  .pio and .vscode, otherwise its too big.
b) the EEZ Studio project using the FLOW method of connecting the pages.

https://github.com/PacciK/EEZ_STUDIO_CYD_01

PacciK, 2024-07-10:
The only useful info I was finding with CYD and EEZ Studio is this helpful chap's channel
 - https://www.youtube.com/watch?v=MJV5cYd6pyM
Where he has two or three videos woking with Arduino ( not PlatformIO)

I have had the simple - no FLOW version working, very similarly to the video above.
But then one has to connect the buttons manually in code, to be able to give fuctions to buttons ?

So I tried using Flow, and when it comes to trying to automatically link the buttons to page turn by Flow, I've had no luck.
I've uploaded the files to my github.

fietser28, 2024-07-10:
If you have LVGL only working from code generated with EEZ Studio then  you can enable flow for platformio by:
1.  adding the eez-framework library to lib_deps in platformio.ini (you don't need to copy file in manually platformio will handle this for you):
```
lib_deps =   https://github.com/eez-open/eez-framework.git
```
2. adding a call to  ui_init() at the end of setup(). This will be called 1 time.
3. adding a call to ui_tick() in the loop() function. This needs to be called frequently (just like the lv_* functions in there)
4. changing the project in EEZ-Studio (or create a new one) with flow support.

I did not test this, but I think this should do it.

PacciK, 2024-07-10:
[image]

@fietser28 - thank you so much for your comment. Its working.
Here's what I did:

I went back to a working LVGL Benchmark demo,
Added the 4 points that you mentioned, and now I have a starting working platform.

Thank you soooooo Much.
Its been a day of trying to get it to work.

PacciK, 2024-07-10:
@mvladic  Martin - ThankYou so much for this amazing work. 
I will now get down to start building something nice.
Please close the issue.

fietser28, 2024-07-10:
Welcome to the world of EEZ-Flow….

For more questions you can join the [Discord](https://discord.gg/SQuWCQEn) community.

closed 2024-07-11 by mvladic