eez-open/eez-framework #12: LVGL9.2 ESP32 IDF build Error

666wsp, 2024-09-13:
Hello! I ported eez-framework to esp32 idf architecture and compiled it with lvgl9.2, idf5.4.0 and it reported an error with two main parts:
### The first part of the error is as follows:
1. error: invalid use of incomplete type 'lv_roller_t' {aka 'struct lv_roller_t'}
  300 | size_t numPages = roller->inf_page_cnt;
The second part reports an error:
2, /eez-framework/src/eez/flow/components/lvgl.cpp:224:63: error: invalid conversion from 'lv_state_t' {aka 'short unsigned int'} to 'lv_obj_ flag_t' [-fpermissive]
  224 | if (booleanValue) lv_obj_add_flag(target, flag);
      | ^~~~
      | ^~~~
      | lv_state_t {aka short unsigned int}

components/eez-framework/src/eez/flow/components/lvgl.cpp:225:52: error: invalid conversion from 'lv_state_t' {aka 'short unsigned int'} to ' lv_obj_flag_t' [-fpermissive]
  225 | else lv_obj_clear_flag(target, flag); | lv_obj_clear_flag(target, flag); lv_obj_flag(target, flag)
      | ^~~~
      | ^~~~
      | lv_state_t {aka short unsigned int}

I can't fix the first part of the error, I don't know where it's caused by not setting the right settings, the second part of the error looks like the book data type doesn't match, I forced the type conversion so it won't report the error, forced the type code: if (booleanValue) lv_obj_add_flag(target, (lv_obj_flag_t)flag); else lv_obj_clear_flag(target, flag); ^~~~ | | lv_state_t {aka short unsigned int}
                    else lv_obj_clear_flag(target, (lv_obj_flag_t)flag);

Translated with DeepL.com (free version)

I compiled the cmake configuration as follows
`if(ESP_PLATFORM)
    file(GLOB_RECURSE SOURCES
        ./src/eez/*.cpp
        ./src/eez/*.c
    )

    idf_component_register(
        SRCS ${SOURCES}
        INCLUDE_DIRS ./src ./src/eez/libs/agg
        REQUIRES lvgl__lvgl
    )

    target_compile_definitions(
        ${COMPONENT_LIB}
        PUBLIC EEZ_FOR_LVGL LV_LVGL_H_INCLUDE_SIMPLE)

    target_compile_options(
        ${COMPONENT_LIB}
        PRIVATE -Wno-error=dangling-pointer)
else()
    cmake_minimum_required(VERSION 3.12)

    include_directories(
        ./src
        ./src/eez/libs/agg
    )

    file(GLOB_RECURSE SOURCES
        ./src/eez/*.cpp
        ./src/eez/*.c
    )

    ADD_LIBRARY(eez-framework STATIC ${SOURCES})

    target_include_directories(eez-framework SYSTEM PUBLIC ./src ./src/eez/libs/agg)
endif()
`
Under lvgl 8.4 I followed this cmake and it works fine

mvladic, 2024-09-13:
You should update eez-framework. Or you don't need to use eez-framework at all if you enable this option in Settings - Build:

[image]

For more information [check this](https://github.com/eez-open/studio/issues/385).

666wsp, 2024-09-13:
> You should update eez-framework.
>
> Or you don't need to use eez-framework at all if you enable this option in Settings - Build: [...]

I did what you did and it still reports an error
[image]
[image]
[image]

mvladic, 2024-09-13:
You don't have the latest version of eez-framework! Your version:

[image]

Latest version:

[image]

https://github.com/eez-open/eez-framework/blob/master/src/eez/flow/components/lvgl.cpp#L222

666wsp, 2024-09-13:
> You don't have the latest version of eez-framework! Your version: [...]

Thank you very much for your help, I fixed it indeed did not download and update to the latest library, it is strange that I use vscode git to switch branches but can not switch to the latest, I directly download can go to the latest, it is possible that my network problem!

Here I got an error compiling as well, I got an error in component->type ! = defs_v3::COMPONENT_TYPE_USER_WIDGET_WIDGET || component->type >= defs_v3::FIRST_DASHBOARD_WIDGET_COMPONENT_TYPE
I added a bracket '()' and it worked, I don't know if it's a bug or just a difference in the compilation system that caused the error.
[image]

mvladic, 2024-09-13:
Your fix is not correct, it should be:

[image]

I fixed this in eez-framework.

666wsp, 2024-09-13:
> Your fix is not correct, it should be: [...]

> Your fix is not correct, it should be: [...]

Okay, I'll change it your way.

mvladic, 2024-09-13:
BTW Normally, this is not an error, but because of `-Werror=parentheses`, warning is promoted to the error.

[image]

We had report of similar case recently. 

In C/C++ following is the same: `a && b || c` and `(a && b) || c` - and both are correct. Operator && has higher priority than ||. By habit, I never use parentheses for such cases.

666wsp, 2024-09-15:
> 顺便说一句，通常这不是一个错误，但是由于`-Werror=parentheses`，警告被升级为错误。
> 
> ![图像](https://private-user-images.githubusercontent.com/449788/367416845-03f3ffd4-df85-4a24-a586-80519d5bd6aa.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjYzMTU4MzcsIm5iZiI6MTcyNjMxNTUzNywicGF0aCI6Ii80NDk3ODgvMzY3NDE2ODQ1LTAzZjNmZmQ0LWRmODUtNGEyNC1hNTg2LTgwNTE5ZDViZDZhYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTE0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkxNFQxMjA1MzdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jZmU3M2Q4ZWUyYmIyYmI0YTM0NmQyMmFjM2IwYmE4MmVhNjg0N2JjZjU3YWFlZWY3ZTQxM2U2OTAwYzlhNjgzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.uWpSVcrUmLJWoHRc4KQXTjSZvb36aFrWoABClOltDEQ)
> 
> 我们最近收到了类似案件的报告。
> 
> 在 C/C++ 中，以下情况相同：`a && b || c`和`(a && b) || c`- 都是正确的。运算符 && 的优先级高于 ||。出于习惯，我从不对此类情况使用括号。
Ok, thank you very much for your reply, I've taken care of it now and can happily play with EEZ Studio!

666wsp, 2024-09-15:
> BTW Normally, this is not an error, but because of `-Werror=parentheses`, warning is promoted to the error. [...]

Ok, thank you very much for your reply, I've taken care of it now and can happily play with EEZ Studio!