eez-open/eez-framework #13: Working partially no flow action

Ju1ien, 2024-10-10:
Hi, I've installed this library on arduino project.
The code complies well and the screen is displayed properly, I see my touch action with buttons changing shape and arc changing position etc.
My problem is when I put a LVGL Flow event on a button/PRESSED. Nothing happens.
I'm using LVGL v9.2.0 and EEZ studio is set as 9.x
I tryed to set the EEZ studio project to 8.x with no change at all. in both cases the display is the same.
I'm maybe missing something, in the generated files, where and how is defined the pressed action of the button? Do I have to code it manually even if it's defined in flow?
Thanks for your help.
julien

mvladic, 2024-10-10:
Do you call `ui_tick()` in the `loop` function, after `lv_timer_handler()` or `lv_task_handler()`? Function `ui_tick` is responsible for the eez flow execution.

mvladic, 2024-10-10:
Great! BTW you can ask these type of questions on our [Discord server](https://discord.gg/37HKvBAt)

closed 2024-10-10 by mvladic