eez-open/eez-framework #14: Comper error on new warning fix

fietser28, 2024-11-16:
With the latest [commit](https://github.com/eez-open/eez-framework/commit/7a01ce78cb06c38d668e4873296b5652ec55b43f) getting getting the the following compile errors:

[image]

Using:
- latest studio commit
- latest eez-framework commit
- Arduino-pico platform on platformio (is using GCC 14.2 toolchain)
all on Mac ARM.

mvladic, 2024-11-16:
Can you try the following?

```
#if __GNUC__ && defined( __has_warning )
#   if __has_warning( "-Wdangling-pointer" )
#       define SUPPRESSING
#       pragma GCC diagnostic push
#       pragma GCC diagnostic ignored "-Wdangling-pointer"
#   endif
#endif

...

#ifdef SUPPRESSING
#   undef SUPPRESSING
#   pragma GCC diagnostic pop
#endif
```

fietser28, 2024-11-16:
Works, no warnings or errors.

[image]

mvladic, 2024-11-16:
Fixed.

commit 3dee6f40 (2024-11-16): Fixed: Compiler error on new warning fix #14
closed 2024-11-16 by mvladic