eez-open/eez-framework #19: LVGL-EEZ Flow does not respond to change screen

aminediab007, 2025-02-27:
**Setup is LVGL 9.2.2, EEZ Studio 0.220, Arduino PlatformIO, ESP32.**

Using LVGL-Flow and the eez-framework, the code compiles without errors and is uploaded to the ESP32, the GUI is displayed on the screen, the button has focus and changes states when pressed, however it does not change screens as expected;  it does change screens  in the run simulator. What am I missing any setting in lv_conf.h file, or something else! Thank you for any assistance.

Here are my settings in EEZ Studio:
- Generate source code for EZ Flow engine(ezz framework) switch is set.

PlatformIo.ini
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = 
	bodmer/TFT_eSPI@^2.5.43
	lvgl/lvgl@^9.2.2
	madhephaestus/ESP32Encoder@^0.11.7
	ipdotsetaf/EZButton@^2.5.0
	https://github.com/eez-open/eez-framework

[lv_conf.txt](https://github.com/user-attachments/files/19002097/lv_conf.txt)

mvladic, 2025-02-27:
Do you call `ui_tick()` after `lv_timer_handler()`?

aminediab007, 2025-02-27:
Indeed, I did not make the call ui_tick() in loop! Everything is working fine now, I am learning this excellent piece of software, and will be supporting you. Thank you for the quick reply :)

closed 2025-02-27 by aminediab007