eez-open/eez-framework #22: Why is there no support for ESP-IDF?

SinglWolf, 2025-03-21:
In my project, I use arduino-ESP32 as a component of the ESP-IDF framework.
Please add in to `CMakeLists.txt`
```cmake
if(ESP_PLATFORM)
# for arduino as component ESP-IDF
get_filename_component(dir ${CMAKE_CURRENT_LIST_FILE} PATH)
FILE(GLOB_RECURSE app_sources ${dir}/src/eez/*.c*)

idf_component_register(SRCS ${app_sources}      
                    INCLUDE_DIRS "src" "src/eez/libs/agg"
                    PRIV_REQUIRES arduino lvgl
)
else()
    cmake_minimum_required(VERSION 3.12)

    include_directories(
        ./src
        ./src/eez/libs/agg
    )

    file(GLOB_RECURSE SOURCES
        ./src/eez/*.cpp
        ./src/eez/*.c
    )

    ADD_LIBRARY(eez-framework STATIC ${SOURCES})

    target_include_directories(eez-framework SYSTEM PUBLIC ./src ./src/eez/libs/agg)
endif()
```

gho1b, 2025-03-21:
I have the same issue. My application looks fine on the screen, but when I try to do something, like change pages, it doesn't work at all. The actions I take aren't being recognized.

I just try to inject logger to the implementation code `eez-flow.cpp` and `lvgl_api.cpp` as follow, the event hitted to lvgl_api.cpp. 
```cpp
extern "C" void flowPropagateValueLVGLEvent(void *flowState, unsigned componentIndex, unsigned outputIndex, lv_event_t *event) {
    ESP_LOGI("EZZ-FLOW", "HITTED");
    lv_event_code_t event_code = lv_event_get_code(event);
    uint32_t code = (uint32_t)event_code;
    void *currentTarget = (void *)lv_event_get_current_target(event);
    void *target = (void *)lv_event_get_target(event);
    void *userDataPointer = lv_event_get_user_data(event);
    int32_t userData = *((int32_t*)(&userDataPointer));
    uint32_t key = 0;
    if (event_code == LV_EVENT_KEY || (event_code == LV_EVENT_VALUE_CHANGED &&
#if LVGL_VERSION_MAJOR >= 9
        lv_obj_check_type((lv_obj_t*)target, &lv_buttonmatrix_class)
#else
        lv_obj_check_type((lv_obj_t*)target, &lv_btnmatrix_class)
#endif
    )) {
        uint32_t *param = (uint32_t *)lv_event_get_param(event);
        key = param ? *param : 0;
    }
    int32_t gestureDir = (int32_t)LV_DIR_NONE;
    if (event_code == LV_EVENT_GESTURE) {
#if LVGL_VERSION_MAJOR >= 9
        lv_indev_wait_release(lv_indev_active());
#else
        lv_indev_wait_release(lv_indev_get_act());
#endif
        gestureDir = (int32_t)lv_indev_get_gesture_dir(
#if LVGL_VERSION_MAJOR >= 9
            lv_indev_active()
#else
            lv_indev_get_act()
#endif
        );
    }
    int32_t rotaryDiff = 0;
#if LVGL_VERSION_MAJOR >= 9
    if (event_code == LV_EVENT_ROTARY) {
        rotaryDiff = lv_event_get_rotary_diff(event);
    }
#endif
    eez::flow::propagateValue(
        (eez::flow::FlowState *)flowState, componentIndex, outputIndex,
        eez::Value::makeLVGLEventRef(
            code, currentTarget, target, userData, key, gestureDir, rotaryDiff, 0xe7f23624
        )
    );
    g_currentLVGLEvent = event;
}
```
```cpp
extern "C" void flowPropagateValueLVGLEvent(void *flowState, unsigned componentIndex, unsigned outputIndex, lv_event_t *event)
{
    ESP_LOGI("LVGL_API", "HITTED");
    lv_event_code_t event_code = lv_event_get_code(event);

    uint32_t code = (uint32_t)event_code;
    void *currentTarget = (void *)lv_event_get_current_target(event);
    void *target = (void *)lv_event_get_target(event);

    void *userDataPointer = lv_event_get_user_data(event);
    int32_t userData = (intptr_t)userDataPointer;

    uint32_t key = 0;
    if (event_code == LV_EVENT_KEY || (event_code == LV_EVENT_VALUE_CHANGED &&
#if LVGL_VERSION_MAJOR >= 9
                                       lv_obj_check_type((lv_obj_t *)target, &lv_buttonmatrix_class)
#else
                                       lv_obj_check_type((lv_obj_t *)target, &lv_btnmatrix_class)
#endif
                                           ))
    {
        uint32_t *param = (uint32_t *)lv_event_get_param(event);
        key = param ? *param : 0;
    }

    int32_t gestureDir = (int32_t)LV_DIR_NONE;
    if (event_code == LV_EVENT_GESTURE)
    {
#if LVGL_VERSION_MAJOR >= 9
        lv_indev_wait_release(lv_indev_active());
#else
        lv_indev_wait_release(lv_indev_get_act());
#endif

        gestureDir = (int32_t)lv_indev_get_gesture_dir(
#if LVGL_VERSION_MAJOR >= 9
            lv_indev_active()
#else
            lv_indev_get_act()
#endif
        );
    }

    int32_t rotaryDiff = 0;
#if LVGL_VERSION_MAJOR >= 9
    if (event_code == LV_EVENT_ROTARY)
    {
        rotaryDiff = lv_event_get_rotary_diff(event);
    }
#endif

    eez::flow::propagateValue(
        (eez::flow::FlowState *)flowState, componentIndex, outputIndex,
        eez::Value::makeLVGLEventRef(
            code, currentTarget, target, userData, key, gestureDir, rotaryDiff, 0xe7f23624));

    g_lastLVGLEvent = *event;
    if (event->user_data)
    {
        g_lastLVGLEvent.user_data = &g_lastLVGLEventUserDataBuffer;
        memcpy(&g_lastLVGLEventUserDataBuffer, event->user_data, sizeof(g_lastLVGLEventUserDataBuffer));
    }
    if (event->param)
    {
        g_lastLVGLEvent.param = &g_lastLVGLEventParamBuffer;
        memcpy(&g_lastLVGLEventParamBuffer, event->param, sizeof(g_lastLVGLEventParamBuffer));
    }
}
```

Did i make mistakes in the implementations?

mvladic, 2025-03-23:
@gho1b Do you call [`ui_tick`](https://github.com/eez-open/studio/wiki/FAQ#how-to-integrate-with-my-existing-cc-project)?

mvladic, 2025-03-23:
@SinglWolf I added changes in CMakeLists.txt for ESP_PLATFORM as you recommended.

gho1b, 2025-03-23:
> [@gho1b](https://github.com/gho1b) Do you call [`ui_tick`](https://github.com/eez-open/studio/wiki/FAQ#how-to-integrate-with-my-existing-cc-project)?

Yes i did, i use the ui_init and ui_tick like this
```c
#include "internal/display.h"

#include "ui/ui.h"

void app_main(void) {
    run_display_2();
    ui_init();
    ui_tick();
}
```

The completed codes in [https://github.com/gho1b/esp-idf-esp32-3248s035c](url)

mvladic, 2025-03-23:
> Yes i did, i use the ui_init and ui_tick like this

You should call `ui_tick` periodically inside some loop. Something like this (see https://docs.lvgl.io/8/porting/timer-handler.html):

```
while(1) {
  lv_timer_handler();
  ui_tick(); // !!!
  my_delay_ms(5);
}
```

But, in your source code I can't find `lv_timer_handler` loop. It is probably somewhere inside `esp_lvgl_port` library. You can use [LVGL timer](https://docs.lvgl.io/master/details/main-modules/timer.html) to call `ui_tick` periodically. Also, I think you should call `ui_init` in the following way:

```
lvgl_port_lock(0);
ui_init();
lvgl_port_unlock();
```

as recommended here https://components.espressif.com/components/espressif/esp_lvgl_port under "LVGL API usage".

gho1b, 2025-03-23:
> > Yes i did, i use the ui_init and ui_tick like this
> 
> You should call `ui_tick` periodically inside some loop. Usually, that loop looks like this (see https://docs.lvgl.io/8/porting/timer-handler.html):
> 
> ```
> while(1) {
>   lv_timer_handler();
>   ui_tick(); // !!!
>   my_delay_ms(5);
> }
> ```
> 
> But, in your source code I can't find `lv_timer_handler` loop. It is probably somewhere inside `esp_lvgl_port` library. Alternatively, you can use [LVGL timer](https://docs.lvgl.io/master/details/main-modules/timer.html) to call `ui_tick` periodically. Also, I think you should call `ui_init` in the following way:
> 
> ```
> lvgl_port_lock(0);
> ui_init();
> lvgl_port_unlock();
> ```
> 
> as recommended here https://components.espressif.com/components/espressif/esp_lvgl_port under "LVGL API usage".

Thanks a lot! Your answer was really helpful.

SinglWolf, 2025-03-23:
@gho1b, I recommend using mutexes if the task with `lv_timer_handler()` is executed in one core and the code in both.

```c
// init
SemaphoreHandle_t xGuiSemaphore = xSemaphoreCreateMutex();

// example core ONE

        if (xSemaphoreTake(xGuiSemaphore, portMAX_DELAY))
        {
            lv_obj_remove_flag(ui_PlayInfo, LV_OBJ_FLAG_HIDDEN);
            lv_obj_add_flag(ui_LOGO, LV_OBJ_FLAG_HIDDEN);
            lv_refr_now(NULL);
            xSemaphoreGive(xGuiSemaphore);
        }
// or

//  task in core TWO
    while (1)
    {
        if (xSemaphoreTake(xGuiSemaphore, portMAX_DELAY))
        {
            lv_timer_handler();
            xSemaphoreGive(xGuiSemaphore);
        }
        vTaskDelay(2);
    }
```

gho1b, 2025-03-23:
@SinglWolf, Appreciate it, i was try without lv_timer_handler, like this one. So far, there is no errors. I will try your approach and thank you to save my days.

```c
#include "internal/display.h"

#include "esp_lvgl_port.h"

#include "ui/ui.h"

void lvgl_task(void *pvParameters)
{
    while (1)
    {
        ui_tick();
        vTaskDelay(pdMS_TO_TICKS(100));
    }
}

void app_main(void)
{
    run_display();
    ui_init();

    xTaskCreate(lvgl_task, "LVGL Task", 4096, NULL, 5, NULL);
}

```

hillsandales, 2025-03-24:
> [@SinglWolf](https://github.com/SinglWolf) I added changes in CMakeLists.txt for ESP_PLATFORM as you recommended.

Adding the proposed changes in the original post causes now causes errors for my build.

```
CMake Error at C:/Users/<user>/esp/v5.3.2/esp-idf/tools/cmake/build.cmake:268 (message):
  Failed to resolve component 'arduino'.
Call Stack (most recent call first):
  C:/Users/<user>/esp/v5.3.2/esp-idf/tools/cmake/build.cmake:310 (__build_resolve_and_add_req)
  C:/Users/<user>/esp/v5.3.2/esp-idf/tools/cmake/build.cmake:305 (__build_expand_requirements)
  C:/Users/<user>/esp/v5.3.2/esp-idf/tools/cmake/build.cmake:607 (__build_expand_requirements)
  C:/Users/<user>/esp/v5.3.2/esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:125 (project)

-- Configuring incomplete, errors occurred!
HINT: The component arduino could not be found. This could be because: component name was misspelled, the component was not added to the build, the component has been moved to the IDF component manager or has been removed and refactored into some other component.
Please look out for component in 'https://components.espressif.com' and add using 'idf.py add-dependency' command.
Refer to the migration guide for more details about moved components.
Refer to the build-system guide for more details about how components are found and included in the build.
cmake failed with exit code 1, output of the command is in the C:\Users\<user>\Documents\gitrepos\waveshare-7in-touch-ha-display\build\log\idf_py_stderr_output_12264 and C:\Users\<user>\Documents\gitrepos\waveshare-7in-touch-ha-display\build\log\idf_py_stdout_output_12264
```

Changing `arduino` to `arduino-esp32` fixes the error.

Espressif seems to put adruino under espressif_arduino-esp32

SinglWolf, 2025-03-24:
> Espressif seems to put adruino under espressif_arduino-esp32

My a folder with "arduino" located in the "components" directory in the root of the project.
[image]
What is your development environment?
I use  IDE VSCode + vscode-esp-idf-extension.

hillsandales, 2025-03-24:
I'm using IDE VSCode with vscode-esp-idf-extension as well. I started with platformio, but changed to esp-idf build system tools for other libraries.

I couldn't get it to build putting things under components/. Adding dependencies with idf.py commands puts them under `managed_components/`.

[image]

Somehow removing the commands from the `if(ESP_PLATFORM)` section above, and configuring other CMakeLists.txt files, I got it to build.

SinglWolf, 2025-03-24:
@hillsandales  I'm still very busy porting my project from `SLS` to `EEZ Studio`. Use the https://github.com/eez-open/studio/issues/761#issuecomment-2746322527 recommendation. 
Later, I will investigate this situation in all possible variants and report here (if I don't forget 😸 ).

haruny, 2025-05-07:
I fixed it : https://github.com/haruny/eez-framework/blame/esp-idf/CMakeLists.txt