eez-open/eez-framework #27: Incorrect Usage of lv_dropdown_set_selected Causing Compiler Error in LVGL 9.3+

jophar, 2025-07-22:
Hello team,

I'm encountering a compiler error related to a malformed call to lv_dropdown_set_selected() in the EEZ Framework. Specifically, the following conditional compilation code appears to be incorrect:

```
#if LVGL_VERSION_MAJOR >= 9 && LVGL_VERSION_MINOR >= 3
    lv_dropdown_set_selected(target, intValue, LV_ANIM_OFF);
#else
    lv_dropdown_set_selected(target, intValue);
#endif
```

According to the LVGL 9.3 API documentation, the correct signature of lv_dropdown_set_selected() **only accepts two parameters**:
[LVGL 9.3 API Reference – lv_dropdown_set_selected](https://docs.lvgl.io/9.3/API/widgets/dropdown/lv_dropdown.html#_CPPv424lv_dropdown_set_selectedP8lv_obj_t8uint32_t)

```
void lv_dropdown_set_selected(lv_obj_t * obj, uint32_t sel);
```

Passing three parameters causes a build failure with LVGL 9.3+. I can manually fix this by adjusting the function call, but since each Studio build regenerates the code, the issue reappears.

Would it be possible to fix this in the code generation logic to avoid manual patching?

Thanks again for your excellent work on the EEZ Studio and framework—it's greatly appreciated!

Best regards,
João

ataweg, 2025-08-14:
It's all well and good that the pull request has been incorporated into lvgl. But the error is still present in the repo.
see https://github.com/eez-open/eez-framework/blob/master/src/eez/flow/components/lvgl.cpp line 270, 789
```
#if LVGL_VERSION_MAJOR >= 9 && LVGL_VERSION_MINOR >= 3
                        lv_dropdown_set_selected(target, intValue, LV_ANIM_OFF);
#else
                        lv_dropdown_set_selected(target, intValue);
#endif
```
EEZ Studio could also use an update for this. The generated code also contains this error.

frankalicious, 2026-01-09:
Since the [fix](e17e7d6) is included in the meantime I think the issue can be closed?